Xinyuan-LilyGO/T7-S3 #1: Esptool Instructions?

jouellnyc, 2022-11-23:
Hello. 

I've just gotten the "LILYGO T7-S3 ESP32-S3 board with 16MB flash, 8MB PSRAM"

Are there steps for using esptool to flash?

I've tried:

```
esptool.py --chip ESP32S3  --port /dev/ttyACM0 write_flash -z 0x0000 ~/esp32/firmware/esp32spiram-20221111-unstable-v1.19.1-660-gc8913fdbf.bin  --force
esptool.py v4.3
Serial port /dev/ttyACM0
Connecting...
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: f4:12:fa:e1:d7:90
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00173fff...
Compressed 1521472 bytes to 985597...
Wrote 1521472 bytes (985597 compressed) at 0x00000000 in 10.7 seconds (effective 1134.4 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

```

On boot gtmkterm will show:

```
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40043ac8
Invalid chip id. Expected 9 read 0. Bootloader for wrong chip?
ets_main.c 329 
ESP-ROM:esp32s3-20210327

```

Most options give errors regarding the chip id, that's why I tried --force:

```
esptool.py --chip ESP32S3   --port /dev/ttyACM0 write_flash -z 0x0000 ~/esp32/firmware/esp32spiram-20221111-unstable-v1.19.1-660-gc8913fdbf.bin 
esptool.py v4.3
Serial port /dev/ttyACM0
Connecting...
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: f4:12:fa:e1:d7:90
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Unexpected chip id in image. Expected 9 but value was 0. Is this image for a different chip model?

A fatal error occurred: /home/john/esp32/firmware/esp32spiram-20221111-unstable-v1.19.1-660-gc8913fdbf.bin is not an ESP32-S3 image. Use --force to flash anyway.

```

lbuque, 2022-11-25:
Where is [this firmware](esp32spiram-20221111-unstable-v1.19.1-660-gc8913fdbf.bin) obtained from?

jouellnyc, 2022-11-25:
Hello. Thanks for the reply. I've tried the firmware here: https://micropython.org/resources/firmware/GENERIC_S3_SPIRAM-20220618-v1.19.1.bin

lbuque, 2022-11-26:
There is currently no micropython firmware for the T7-S3.

you can try this [firmware](https://github.com/Xinyuan-LilyGO/lilygo-micropython/suites/8922903601/artifacts/409089943).

jouellnyc, 2022-11-26:
Thank you!

I have tried:

```
esptool.py --chip ESP32S3  --port /dev/ttyACM0 write_flash -z 0x0000 LilyGo-MicroPython_esp32s3_LILYGO_T-DisplayS3_MPY-d7919ea71_IDF-1b16ef6cfc_B20221024_a07409ec59.bin 

esptool.py v4.3
Serial port /dev/ttyACM0
Connecting...
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: f4:12:fa:e1:d7:90
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x0017cfff...
Compressed 1557328 bytes to 991536...
Wrote 1557328 bytes (991536 compressed) at 0x00000000 in 10.7 seconds (effective 1162.2 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

```
It works now and I use it with Thonny IDE. 

Thank you @lbuque !!!

lbuque, 2022-11-26:
This is the micropython [firmware](https://github.com/Xinyuan-LilyGO/lilygo-micropython/suites/9514697728/artifacts/451759541) for T7-S3.

This is the board micropython configuration code of T7-S3.

https://github.com/Xinyuan-LilyGO/lilygo-micropython/tree/master/target/esp32s3/boards/LILYGO_T7-S3

For subsequent micropython issues related to this board, please move to https://github.com/Xinyuan-LilyGO/lilygo-micropython/issues.

This issue will be closed.

closed 2022-11-26 by lbuque

jouellnyc, 2022-11-26:
Thanks again for the information and response.