Xinyuan-LilyGO/T7-S3 #5: Touch Pins

disruptivepatternmaterial, 2023-01-30:
Has anyone been able to get the touch pins to work? I have tried many ways including just reading all of them and touching them one at a time. None of them register. On my other esp32s3 modules they "just work." So I think something is broken with this board. I'd love to be wrong and find out the secret to using them here.

maxdd, 2023-06-10:
IO2, which is TOUCH2, is used to sense the battery so that cannot be used.
I'm not an expert with TOUCH but if you say you should touch them individually and expect them to register and they dont, maybe it's a PU/PD issue?

jalmince, 2023-06-13:
for me it works !

touch GPIO04 and touchRead(4) increases about 10 fold

Touch: 27771
Touch: 28479
Touch: 29602
Touch: 216505  (touched)
Touch: 321552  (touched)
Touch: 360484  (touched)
Touch: 320490  (touched)
Touch: 23546
Touch: 23466
Touch: 23474
```
#include <Arduino.h>
#define BAT_ADC        2
#define LED_PIN        17

void setup() {
  Serial.begin(115200);
  delay(1000);  
  Serial.println("ESP32 Touch Test");
}

void loop() {
  Serial.print("Touch: ");
  Serial.println(touchRead(4));  // get touch value on GPIO 4
  delay(1000);
}
```