Xinyuan-LilyGO/T7-S3 #11: Where to find I2C pins

hcl-83, 2024-09-02:
Hello,
i want to connect MPU-9250 sensor but dont see any of I2C pins on chart : 

[image]

Can i use any of gpio 45-48 ??

cpocol, 2025-06-21:
Look for Wire_SCL and Wire_SCA (pins 22, 21)

tomhaines, 2026-04-27:
>    1.
>
>    Right, the diagram posted by @hcl-83 <https://github.com/hcl-83> is
>    wrong.
>    2.
>
>    In one of my projects, I have an MPU6050 (I need it as an
>    accelerometer).
>
> In the global space I have:
> image.png (view on web)
>
> and in setup() I have:
> image.png (view on web)

Thank you. Which physical pins did you connect to?